sirpdboy/openwrt #1: 硬盘挂载

wjhcuicui, 2020-07-07:
硬盘怎么都挂载不上去，是不是bug

sirpdboy, 2020-07-11:
要用EXT4格式，NTFS不支持。

jimmyzhou310, 2020-09-08:
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.098981] squashfs: Unknown parameter 'discard'
Tue Sep  8 16:36:21 2020 kern.info kernel: [   37.101301] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.103042] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Tue Sep  8 16:36:21 2020 kern.info kernel: [   37.104943] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x66fdb73a)
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.106472] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.108861] fuseblk: Unknown parameter 'discard'
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.110652] exFAT-fs (sda128): invalid boot record signature
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.112155] exFAT-fs (sda128): failed to read boot sector
Tue Sep  8 16:36:21 2020 kern.err kernel: [   37.113574] exFAT-fs (sda128): failed to recognize exfat type

leimac2020, 2020-09-09:
可以挂载而且我还挂了两块

zydfq, 2020-10-31:
NTFS 格式有同样问题
10-26没问题
10-28养老版有问题
10-29有问题 
估计是不识别NTFS格式。造成磁盘挂载不上。

zydfq, 2020-11-03:
郁闷
11-02版本问题依旧
目前只能用10.26版本
期待解决问题
另外
11-02应用控制很实用
11-02磁盘管理中创建菜单无内容。

zydfq, 2020-11-04:
郁闷
11-04版本硬盘问题依旧，无法共享、FTP......

zydfq, 2020-11-04:
郁闷
11-04版本硬盘问题依旧，无法共享、FTP......

硬件环境：
CPU：Intel(R) Xeon(R) CPU E31230 @ 3.20GHz
内存：4G
网卡：Intel 光卡
硬盘：8G固态（OpenWrt系统）+ 1.5T 机械硬盘NTFS格式（数据共享）

szkifen, 2020-12-05:
我就回到10月26号

closed 2025-05-14 by sirpdboy