sirpdboy/openwrt #10: 多线多播模式下 WAN6 端口造成 CPU占用100% 用户无法上网

zydfq, 2020-08-28:
截取系统日志：
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:00 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): cat: write error: Broken pipe
Fri Aug 28 17:47:01 2020 daemon.notice netifd: wan6 (14192): Command failed: Request timed out

MLST10019, 2020-08-29:
我刚从825的ipv6大全版更新到828大全版。
单线多拨，保留配置更新。
更新完成后
所有的eth0～4后面都出现了wan6
（正常来说只有wan口后面有wan6的）
更新后lan后面也有wan6了
这导致拨不上了

sirpdboy, 2020-09-11:
删除WAN6。新版本已经取消多网卡绑定了。

closed 2025-07-24 by sirpdboy