sirpdboy/openwrt #11: 有办法解决 - squashfs: Unknown parameter 'discard'

jimmyzhou310, 2020-09-08:
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.098981] squashfs: Unknown parameter 'discard'
Tue Sep 8 16:36:21 2020 kern.info kernel: [ 37.101301] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.103042] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Tue Sep 8 16:36:21 2020 kern.info kernel: [ 37.104943] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x66fdb73a)
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.106472] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.108861] fuseblk: Unknown parameter 'discard'
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.110652] exFAT-fs (sda128): invalid boot record signature
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.112155] exFAT-fs (sda128): failed to read boot sector
Tue Sep 8 16:36:21 2020 kern.err kernel: [ 37.113574] exFAT-fs (sda128): failed to recognize exfat type

humanxp, 2021-09-02:
同样的问题
[   37.863946] squashfs: Unknown parameter 'discard'
[   37.866988] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   37.867291] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   37.867846] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
[   37.868141] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   37.871348] [EXFAT] trying to mount...

humanxp, 2021-09-03:
应该是openwrt-x86-64-generic-squashfs-combined-efi文件
换成openwrt-x86-64-generic-squashfs-combined没有efi文件试试

Fyred, 2023-01-31:
> 应该是openwrt-x86-64-generic-squashfs-combined-efi文件 换成openwrt-x86-64-generic-squashfs-combined没有efi文件试试

你好，请问这个问题最后解决了吗？谢谢