sirpdboy/openwrt #31: overlay分区被写爆了，导致某些插件配置不全无法使用

Delayless, 2020-10-21:
[image]

现象如上图，`/etc/config/vssr`配置文件不完整，从中间截断，后半部分由于内存不足丢失了。
我没有额外安装任何插件，就是直接用您的固件，改改配置使用的，而且使用的插件也就那么五六个，并没有全部使用。
然后我用df查看了下磁盘情况，发现overlay完全满了，然后我进到/overlay目录用du查看了下已用空间只有200M，如图：
[image]

分析到这里，就不知道该怎么办了，望帮忙解决一下，谢谢您。

fsg1020, 2020-10-25:
我和你一样，碰到这个问题，开始了也没发现是AD造成的，现在每次关注/overlay使用，多了，就到AD中清理下。我挂载了一个分区，但貌似没成功。

sirpdboy, 2020-10-27:
空间不够，可以扩展/overlay 。或者清除空间。

Delayless, 2020-10-27:
> 空间不够，可以扩展/overlay 。或者清除空间。

嗯，不过我觉得可以在源码添加保护机制，不至于配置文件不完整导致插件不能用（我都是又建了一个虚拟机提取出原始配置文件才复原的）

closed 2022-01-09 by Delayless