sirpdboy/openwrt #51: J1900主机刷这个固件 网速很慢，之前旧版本的固件 宽带轻松跑满

Ja1lbreakra1n, 2020-12-09:


asnon, 2020-12-10:
之前那个版本可以跑满？

Ja1lbreakra1n, 2020-12-10:
我刷的别人的一个固件，很老了

 季羿臣
品牌策划  Brand Planner
手机 Mob：(+86)18061813030

启东黄金海滩景区
QiDong Golden Beach Scenic Area
江苏省启东市圆陀角旅游度假区黄金海滩景区（G345线最东端）（邮编：226200）
Golden Beach, Yuantuojiao Scenic Spot, Qidong 226200, P.R. China
电话 Tel：(86 513)6991 9888
网站 Web：www.huangjinhaitan.cn

Ja1lbreakra1n, 2020-12-10:
我晚上回去把之前那个固件发你看看

 季羿臣
品牌策划  Brand Planner
手机 Mob：(+86)18061813030

启东黄金海滩景区
QiDong Golden Beach Scenic Area
江苏省启东市圆陀角旅游度假区黄金海滩景区（G345线最东端）（邮编：226200）
Golden Beach, Yuantuojiao Scenic Spot, Qidong 226200, P.R. China
电话 Tel：(86 513)6991 9888
网站 Web：www.huangjinhaitan.cn

Ja1lbreakra1n, 2020-12-10:
这是我刷的老版本的固件
 季羿臣
品牌策划  Brand Planner
手机 Mob：(+86)18061813030

启东黄金海滩景区
QiDong Golden Beach Scenic Area
江苏省启东市圆陀角旅游度假区黄金海滩景区（G345线最东端）（邮编：226200）
Golden Beach, Yuantuojiao Scenic Spot, Qidong 226200, P.R. China
电话 Tel：(86 513)6991 9888
网站 Web：www.huangjinhaitan.cn

Ja1lbreakra1n, 2020-12-12:
[](url)

> 之前那个版本可以跑满？

https://drive.google.com/file/d/1P3WUBePVA7wpuUiiaMAazNQADTceQabz/view?usp=sharing
这个是我刷过唯一一个可以正常跑满宽带的固件
硬件是j1900 4gb 双网口 螃蟹r8111的千兆网卡，直接刷的uefi固件未使用虚拟机

closed 2025-07-24 by sirpdboy