sirpdboy/openwrt #58: 系统日志一堆错误

howqua, 2020-12-26:
是我把事情搞砸了吗？

Sat Dec 26 13:43:14 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:16 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:18 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:20 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:22 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:24 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:26 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:28 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Dec 26 13:43:30 2020 daemon.err netdata[3799]: PROCFILE: Cannot open file '/proc/sysvipc/shm'

MCydia, 2020-12-29:
netdata 也就是实时监控等报错，你去关闭就好了

在netdata的配置文件里把err log关掉
    debug log = none
    error log = none
    access log = none

1. 编辑 /etc/netdata/netdata.conf
2. 添加下面的内容
【plugin:proc:ipc】
 semaphore totals = no
3. 重启netdata 插件

sirpdboy, 2021-01-06:
新版本已经处理好了

closed 2025-07-24 by sirpdboy