sirpdboy/openwrt #65: PPTP client无法安装

938134, 2021-01-02:
root@OpenWrt:~# opkg install ppp-mod-pptp --force-depends
Installing ppp-mod-pptp (2.4.8.git-2020-10-03-2) to root...
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/base/ppp-mod-pptp_2.4.8.git-2020-10-03-2_x86_64.ipk
Installing kmod-pptp (5.4.86-1) to root...
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/x86/64/packages/kmod-pptp_5.4.86-1_x86_64.ipk
Configuring kmod-pptp.
Configuring ppp-mod-pptp.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pptp:
 *      kernel (= 5.4.86-1-1cc178c1f8d6c85a5513a864a5d45c6e)
root@OpenWrt:~# Cannot satisfy the following dependencies for ppp-mod-pptp:
-ash: Cannot: not found

sirpdboy, 2021-01-06:
内核不匹套

938134, 2021-01-09:
能修复吗

closed 2025-07-24 by sirpdboy