sirpdboy/openwrt #67: OpenWrt S2021.01.07 Ipv6 by Sirpdboy 错误报告

Ahuababa, 2021-01-08:
/usr/lib/lua/luci/util.lua:627: Unable to establish ubus connection
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/util.lua:627: in function 'ubus'
	/usr/lib/lua/luci/dispatcher.lua:172: in function 'session_retrieve'
	/usr/lib/lua/luci/dispatcher.lua:373: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

Ahuababa, 2021-01-09:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/util.lua:627: Unable to establish ubus connection
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

sirpdboy, 2021-01-20:
保留配置刷机没弄好  ，重新写盘就行。

closed 2025-07-24 by sirpdboy