sirpdboy/openwrt #84: koolddns针对dnspod需要修改输入方式为token

AlfredZuo, 2021-01-17:
dnspod国内版在使用时使用Token进行鉴权，Token获取方式如下
https://console.dnspod.cn/account/token

但是在1月14日的kooldns配置中，只能按照Access Key与Signature的方式鉴权，这个方式无法通过dnspod国内版的鉴权
需要更改鉴权参数为Token

AlfredZuo, 2021-01-17:
[image]

AlfredZuo, 2021-01-17:
[image]
另外，检查IP是不是可以默认换为ip.3322.com

sirpdboy, 2021-01-20:
本人时间和精力 有限。源码研究太费时间，你有时间研究解决方案，一起努力 将源码修复。

sangzhesvh, 2021-01-23:
koolddns对dnspod用的就是token，在Accesskey处填入token，Signature处填ID。但是原码的更新方式有问题，虽然显示更新了，实际没有更新，而且能正确显示未更新的旧IP

sirpdboy, 2021-02-10:
2021.2.10号版本修正此问题了。

Se7enMuting, 2021-02-11:
还是报错，但是可以更新。我测试：直接在DNSpod里把IP改了，插件会直接新增一条正确的A记录，而不是把那条错的A记录改掉。。。