sirpdboy/openwrt #107: 网络共享

zydfq, 2021-02-28:


chenghaitao, 2021-03-02:
[global]
	netbios name = |NAME|
	server string = |DESCRIPTION|
	workgroup = |WORKGROUP|
	interfaces = |INTERFACES|
	bind interfaces only = yes
	ipc timeout = 20
	deadtime = 15
	map to guest = Bad User
	smb2 max read = 64K
	smb2 max write = 64K
	smb2 max trans = 64K
	cache read buffers = no
	cache trans buffers = no

勾选自动共享，可浏览，允许匿名访问，使用root，点击应用就可以了，没有你上面说的问题。

zydfq, 2021-03-03:
1）共享名称不支持中文。
请您建立共享时在名字栏目下以中文命名共享，比如“共享”   “/mnt/sdb1”  

2）
在win10下通过共享路径“192.168.1.1\sj\"粘贴一压缩文件”XXXX NNN.RAR“然后关闭窗口；再次打开路径 “192.168.1.1\sj\"将文件”XXXX NNN.RAR“复制到win10任何路径，或右键擦看文件”XXXX NNN.RAR“。windows文件资源管理器停止响应。

重启电脑，或重启windows文件资源管理器。用"FTP://192.168.1.1"登录打开 “/mnt/sdb1”  路径，拷贝、删除、移动文件”XXXX NNN.RAR“；正常。

您发给我上面的代码，与您OpenWrt中的一样。

zydfq, 2021-03-12:
2021.03.09 版本 网络共享问题解决

closed 2025-07-24 by sirpdboy