sirpdboy/openwrt #130: 请求大大不要每次更新都变更插件内容

zydfq, 2021-04-22:
作为您作品的粉丝，请求大大不要每次更新都变更插件内容。频繁变更功能插件，我们真的没法跟进更新。很多功能我们是拿来用的，不是单纯的测试啊！刚用上上个版本的可道云备份了手机，这个版本又把可道云给卸了。郁闷！

20号IPV6大全版本，和介绍的功能不一样。另外取消了可道云。

BYPASS                      感觉还不错保留。
koolddns                   比较好用的DDNS插件，兼容性和稳定性都不错建议保留。
SmartDNS                 感觉功效不明显，有些鸡肋。建议去掉！
KMS                           关键功能，现在很稳定，保留。
AdGuard Home         虽然设置很麻烦但是同时具备屏蔽广告和DNS优化功能，建议保留。
广告过滤大师 Plus+  不明白楼主为什么留着它，而不用更简单好用的广告过滤插件？
Ksmbd                       至今不兼容中文共享目录！samba功能强大支持中文共享目录，早期版本楼主已经把samba搞定了，不知为什么又放弃samba了？
多线多拨                    关键功能，一定保留。
miniDLNA                  稳定的流媒体服务器，用电视看大片全靠这个插件，保留！
DiskMan 磁盘管理     20号IPV6大全版本该插件有问题，作为核心功能插件请尽快修复，并保留！
可道云                       可视化APP云存储，建议保留，并进一步开发。
下载类插件                请尽量稳定一两个插件。

杰作！一定是精品插件的结合体。

HANSCOM123, 2021-04-27:
您可以不更新啊

Dante-github, 2021-04-27:
之前提的好几个建议，大大都采纳了，虽然0420的版本，bypass插件有问题无法启动，但还是很感激。估计是在bypass使用过程中无法和samrtDNS很好的兼容，所以0426版本又去掉了，但还是希望能够调好之后再加回来。smartDNS的配置貌似配置了会和其他功能冲突导致无法上网，也不知道是不是我没配置好。
希望越来越好

zydfq, 2021-05-09:
> 您可以不更新啊
更新大大的作品，回馈测试信息，提合理化建议在我的理解里是对作者的回报和尊重。希望理解。

dysobo, 2021-06-24:
> > 您可以不更新啊
> > 更新大大的作品，回馈测试信息，提合理化建议在我的理解里是对作者的回报和尊重。希望理解。

哈哈，我也经常被这么建议，其实倒不是恶意。他就是建议路由器版本，如无必要，不需要经常升级~