sirpdboy/openwrt #141: 大神我的酸乳+怎么显示内部错误500鸭

592174937, 2021-05-28:
恩山上这个帖子里下的 白嫖精简版
https://www.right.com.cn/forum/thread-4047549-1-1.html

592174937, 2021-05-28:
500 内部服务器错误
对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:74: Failed to load template 'openclash/myip'.
Error while parsing template '/usr/lib/lua/luci/view/openclash/myip.htm':
No such file or directory
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

sirpdboy, 2021-06-17:
没空间了。重置重新设置