sirpdboy/openwrt #148: 路由器间隔性断网。小白不会看日志，有没有大佬帮看一下呀。

710511985, 2021-06-12:
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: netdata initialization completed. Enjoy real-time performance monitoring!
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8814
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8814 to PLUGIN[diskspac
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8815
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8815 to PLUGIN[tc]
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: CONFIG: cannot load user exporting config '/etc/netdata/exporting.conf'. Will try the stock version.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8822
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8822 to WEB_SERVER[stat
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8824
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: starting worker 2
Sat Jun 12 18:44:04 2021 daemon.err netdata[8368]: This agent doesn't have ACLK.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8817
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8817 to PLUGIN[idlejitt
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/system.idlejitter/main.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8819
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8819 to BACKENDS
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: cleaning up...
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread with task id 8819 finished
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8812
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8812 to PLUGIN[proc]
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8824 to HEALTH
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8833
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8833 to WEB_SERVER[stat
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8818
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8818 to STATSD
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/system.idlejitter/min.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/system.idlejitter/max.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8823
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8823 to PLUGINSD
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/system.idlejitter/average.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8837
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8837 to PLUGINSD[ioping
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8838
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8838 to PLUGINSD[perf]
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: starting worker 3
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: starting worker 4
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8842
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8842 to WEB_SERVER[stat
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8841
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8841 to WEB_SERVER[stat
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: No connector instances to activate
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: EXPORTING: no exporting connectors configured
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: cleaning up...
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread with task id 8821 finished
Sat Jun 12 18:44:04 2021 daemon.err netdata[8368]: STATSD: invalid type 'avg' at line 56 of file '/usr/lib/netdata/conf.d/statsd.d/k6.conf'. Using 'last'.
Sat Jun 12 18:44:04 2021 daemon.err netdata[8368]: STATSD: invalid type 'avg' at line 102 of file '/usr/lib/netdata/conf.d/statsd.d/k6.conf'. Using 'last'.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: connected to '/usr/lib/netdata/plugins.d/ioping.plugin' running on pid 8840
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/main.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: connected to '/usr/lib/netdata/plugins.d/perf.plugin' running on pid 8839
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/gauges.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/counters.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: thread created with task id 8851
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: set name of thread 8851 to STATSD_COLLECTO
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: STATSD collector thread started with taskid 8851
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: POLLFD: LISTENER: listening on 'udp:[::1]:8125'
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/timers.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/meters.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/histograms.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_metrics/sets.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/main.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/gauges.db.
Sat Jun 12 18:44:04 2021 daemon.info netdata[8368]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/counters.db.
[... 554 lines cut ...]
Sat Jun 12 18:52:04 2021 daemon.err uhttpd[3660]: 	[C]: in function 'assert'
Sat Jun 12 18:52:04 2021 daemon.err uhttpd[3660]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Sat Jun 12 18:52:04 2021 daemon.err uhttpd[3660]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute function dispatcher target for entry '/admin/nlbw/usage'.
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: The called action terminated with an exception:
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute template dispatcher target for entry '/admin/nlbw/usage/details'.
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: The called action terminated with an exception:
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'wrtbwmon'.
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: A runtime error occured: [string "/usr/lib/lua/luci/view/wrtbwmon.htm"]:50: attempt to concatenate local 'db' (a nil value)
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: stack traceback:
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: 	[C]: in function 'assert'
Sat Jun 12 18:52:06 2021 daemon.err uhttpd[3660]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
[... cut ...]
Sat Jun 12 18:52:21 2021 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat Jun 12 18:52:22 2021 user.notice wrtbwmon: Restart for ifup of wan_6 (pppoe-wan)
Sat Jun 12 18:52:43 2021 daemon.notice ddnsto[20170]: main accept error count=0 wait_cnt=0
Sat Jun 12 18:52:48 2021 daemon.notice ddnsto[7853]: check reboot failed, ret=-1
Sat Jun 12 18:53:27 2021 daemon.notice ddnsto[29559]: client init ok username=e39e4c39-5cdf-4f70-afcc-fc4401b00fba routerId=00e04c68e244 version=0.2.9

710511985, 2021-06-12:
在18点51分左右断的。麻烦看一下。

710511985, 2021-06-12:
@sirpdboy

sirpdboy, 2021-06-17:
Sat Jun 12 18:44:51 2021 daemon.err pppd[11985]: Interface eth1 has MTU of 1480 -- should be at least 1500.
Sat Jun 12 18:44:51 2021 daemon.err pppd[11985]: This may cause serious connection problems.
Sat Jun 12 18:44:52 2021 daemon.info avahi-daemon[4282]: Interface ztks5z7a7n.IPv6 no longer relevant for mDNS.
Sat Jun 12 18:44:52 2021 daemon.info avahi-daemon[4282]: Leaving mDNS multicast group on interface ztks5z7a7n.IPv6 with address fe80::b8f4:76ff:fee9:2e41.
Sat Jun 12 18:44:52 2021 user.err wsdd2[9390]: error: wsdd-mcast-v4: wsd_send_soap_msg: send: Network unreachable
Sat Jun 12 18:44:52 2021 daemon.info avahi-daemon[4282]: Withdrawing address record for fe80::b8f4:76ff:fee9:2e41 on ztks5z7a7n.
Sat Jun 12 18:44:53 2021 daemon.err zerotier-one[12210]: sendto: Network unreachable
Sat Jun 12 18:44:55 2021 daemon.info avahi-daemon[4282]: Joining mDNS multicast group on interface ztks5z7a7n.IPv6 with address fe80::1c02:b7ff:fe10:ac5b.
Sat Jun 12 18:44:55 2021 daemon.info avahi-daemon[4282]: New relevant interface ztks5z7a7n.IPv6 for mDNS.
Sat Jun 12 18:44:55 2021 daemon.info avahi-daemon[4282]: Registering new address record for fe80::1c02:b7ff:fe10:ac5b on ztks5z7a7n..
Sat Jun 12 18:44:58 2021 kern.info kernel: [ 90.943018] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Down
Sat Jun 12 18:44:58 2021 kern.info kernel: [ 90.947266] br-lan: port 2(eth2) entered disabled state
Sat Jun 12 18:44:58 2021 daemon.notice netifd: Network device 'eth2' link is down
Sat Jun 12 18:44:58 2021 kern.info kernel: [ 91.095015] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Down
Sat Jun 12 18:44:59 2021 daemon.notice netifd: Network device 'eth1' link is down
Sat Jun 12 18:44:59 2021 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Jun 12 18:44:59 2021 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Jun 12 18:44:59 2021 daemon.notice netifd: Interface 'wan6' is now down
Sat Jun 12 18:45:04 2021 daemon.notice netifd: Interface 'wan' is now down

你的这日志 不断提示WAN 断开连接。。。换网线或者 重置一下路由试试。