sirpdboy/openwrt #154: 6.18版本

zydfq, 2021-06-18:
感觉上网反应速度快于上一版本 

但是发现一下问题
1）导入上一版本备份文件后，密码错误无法登录。
2）koolddns 远程解析IP 反馈 {error}
3）防火墙 - 端口转发 - 匹配规则 栏 反馈 { wan 源于 uci: Entry not found }

zydfq, 2021-06-18:
> 感觉上网反应速度快于上一版本
> 
> 但是发现一下问题
> 1）导入上一版本备份文件后，密码错误无法登录。
> 2）koolddns 远程解析IP 反馈 {error}   但是解析正常！
> 3）防火墙 - 端口转发 - 匹配规则 栏 反馈 { wan 源于 uci: Entry not found }

eggcn, 2021-06-19:
和楼主一样的问题，同样的版本同样是导入上一版本备份文件，换了浏览器啥的都都没用，就是提示无效的用户名和/或密码！请重试。

eggcn, 2021-06-19:
@sirpdboy 呼叫老大给个解决方案

chfjxmtable, 2021-06-19:
备份恢复出现密码错误，但在SSH上依然是password这个密码

eggcn, 2021-06-19:
> 备份恢复出现密码错误，但在SSH上依然是password这个密码

受到你的启发，我用ssh登陆可以，尝试使用passwd命令重置密码，修改密码结果无效，还是只能用之前备份的密码登陆ssh。

chfjxmtable, 2021-06-19:
我只有使用firstboot删除设置内容，然后再重启来手动设置，但重新做一遍softerther vpn设置真得很烦人。

chfjxmtable, 2021-06-21:
奇怪，这次连softether vpn设置时发现居然服务器无法连接

sirpdboy, 2021-07-31:
5.4内核升级到5.10的内核就会这样，建议不要保留配置升级，也不要恢复5.4的内核的备份。。

closed 2025-07-24 by sirpdboy