sirpdboy/openwrt #228: 出现一下问题不知如何解决？

icons88, 2021-11-04:
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:443 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:80 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:443 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Thu Nov  4 11:21:03 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:80 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:443 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:80 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:443 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Thu Nov  4 11:21:04 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:80 failed (98: Address in use)
Thu Nov  4 11:21:05 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Thu Nov  4 11:21:05 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:443 failed (98: Address in use)
Thu Nov  4 11:21:05 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Thu Nov  4 11:21:05 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: bind() to [::]:80 failed (98: Address in use)
Thu Nov  4 11:21:05 2021 daemon.err nginx[13453]: 2021/11/04 11:21:03 [emerg] 13453#0: still could not bind()

fsyxatcq, 2021-11-06:
80、443端口被运营商封了，不能够绑定

icons88, 2021-11-07:
明白了，谢谢解答！

closed 2025-07-24 by sirpdboy