sirpdboy/openwrt #233: esxi直通Intel i350-t2网卡，网卡会无限重启

visvii, 2021-11-15:
[image]

ericshunhawk, 2021-11-16:
同问题 升级到esxi7.0就ok了

visvii, 2021-11-16:
> 同问题升级到esxi7.0就ok了

升级后确实解决了，感谢

ericshunhawk, 2021-11-20:
> > 同问题升级到esxi7.0就ok了
> 
> 
> 
> 升级后确实解决了，感谢

有没有运行一段时间之后再次反复重启的问题？

visvii, 2021-11-20:
> > > 同问题升级到esxi7.0就ok了
> > 
> > 
> > 升级后确实解决了，感谢
> 
> 有没有运行一段时间之后再次反复重启的问题？

暂时没发现，只用了会7.0感觉不好用又降回6.7了；后面找到个6.7直通i350网卡不会重启的固件：https://www.luyouwang.net/4681.html，用了3天挺稳定

ericshunhawk, 2021-11-21:
> > > > 同问题升级到esxi7.0就ok了
> 
> > > 
> 
> > > 
> 
> > > 升级后确实解决了，感谢
> 
> > 
> 
> > 有没有运行一段时间之后再次反复重启的问题？
> 
> 
> 
> 暂时没发现，只用了会7.0感觉不好用又降回6.7了；后面找到个6.7直通i350网卡不会重启的固件：https://www.luyouwang.net/4681.html，用了3天挺稳定

感谢 我去看看

ericshunhawk, 2021-11-25:
> > > > 同问题升级到esxi7.0就ok了
> 
> > > 
> 
> > > 
> 
> > > 升级后确实解决了，感谢
> 
> > 
> 
> > 有没有运行一段时间之后再次反复重启的问题？
> 
> 
> 
> 暂时没发现，只用了会7.0感觉不好用又降回6.7了；后面找到个6.7直通i350网卡不会重启的固件：https://www.luyouwang.net/4681.html，用了3天挺稳定

我还是用的最新内核 把主板网口建了个虚拟交换机单独给了lede的wan口 然后直通了一个i350网口给lede的lan口 貌似也没问题了 pppoe测速也能达到千兆

visvii, 2021-11-29:
> 

有试过用虚拟网卡做wan口，多拨后能稳定吗，有没频繁掉线

visvii, 2021-11-29:
> 我还是用的最新内核 把主板网口建了个虚拟交换机单独给了lede的wan口 然后直通了一个i350网口给lede的lan口 貌似也没问题了 pppoe测速也能达到千兆

我最近在测试PVE，直通网卡比esxi方便点，能查看cpu温度

ericshunhawk, 2021-11-30:
> > 
> 
> 
> 
> 有试过用虚拟网卡做wan口，多拨后能稳定吗，有没频繁掉线

没测试过多播 我已经千兆光纤了 内网也只有千兆 多拨没多大意义 上传100mb够了

ericshunhawk, 2021-11-30:
> 我最近在测试PVE，直通网卡比esxi方便点，能查看cpu温度

啥板子啊

visvii, 2021-11-30:
> 

工控机，那种集成的主板，CPU是i5 8250u

closed 2025-07-24 by sirpdboy