sirpdboy/openwrt #280: iKoolProxy无法开启

sofs2005, 2022-02-05:
我是从1月1号的版本升级到2月1号的版本的，上个版本是正常的，2月1号的版本无法开启，我配置都没动过，关键是也不知道哪里出错了，页面上的状态是红绿交替，也不知道到底有没有运行。

sofs2005, 2022-02-05:
补充一下，我用的是大全版

pengpeng21, 2022-02-06:
问过ikoolproxy的维护人，ikoolproxy目前不支持5.10版本

fw567, 2022-02-07:
ikoolproxy本质就是koolproxy，koolproxy不支持5.10内核，ikoolproxy当然也不支持

sofs2005, 2022-02-09:
那看来大佬要重新招一个插件了。。。

sofs2005, 2022-02-14:
情人节版谁试过去广告好用不？不行我打算用回元旦版了

refined-fish, 2022-02-16:
> 

看更新日志里面没有修复这个问题且任然是5.10内核，也就是说依旧存在这个问题，另外借楼问问各位朋友有没有好的拦截规则，我装了证书https拦截但是国内各大视频网站仍然有广告，不知道是规则问题还是这些广告无法屏蔽？

sofs2005, 2022-02-17:
我刷了情人节版，一样的问题。刷回元旦版了，这个去广告还是比较牛逼的，各个视频广告都能干掉，爽啊。大家有新核心能用的去广告的时候记得喊我一声

refined-fish, 2022-02-19:
> 我刷了情人节版，一样的问题。刷回元旦版了，这个去广告还是比较牛逼的，各个视频广告都能干掉，爽啊。大家有新核心能用的去广告的时候记得喊我一声

真的么，我怎么用的也是元旦版但是爱奇艺啥的广告还在啊

sofs2005, 2022-02-20:
> > 我刷了情人节版，一样的问题。刷回元旦版了，这个去广告还是比较牛逼的，各个视频广告都能干掉，爽啊。大家有新核心能用的去广告的时候记得喊我一声
> 
> 真的么，我怎么用的也是元旦版但是爱奇艺啥的广告还在啊

电视版的各种app默认都可以过滤，电脑上我试了下需要装证书，没证书的话https的视频过滤不掉

sofs2005, 2022-03-10:
老大改回老内核了，没问题了

closed 2022-03-10 by sofs2005