sirpdboy/openwrt #283: very poor speed inside the network

den5is5, 2022-02-12:
iperf3.exe -c 192.168.1.1
Connecting to host 192.168.1.1
[  4] local 127.0.0.1  connected to 192.168.1.1
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   104 MBytes   873 Mbits/sec
[  4]   1.00-2.00   sec  92.0 MBytes   772 Mbits/sec
[  4]   2.00-3.00   sec  96.0 MBytes   806 Mbits/sec
[  4]   3.00-4.00   sec  92.0 MBytes   771 Mbits/sec
[  4]   4.00-5.01   sec  94.1 MBytes   785 Mbits/sec
[  4]   5.01-6.01   sec  93.9 MBytes   788 Mbits/sec
[  4]   6.01-7.01   sec  92.9 MBytes   779 Mbits/sec
[  4]   7.01-8.00   sec  93.2 MBytes   787 Mbits/sec
[  4]   8.00-9.01   sec  93.9 MBytes   779 Mbits/sec
[  4]   9.01-10.00  sec  92.9 MBytes   787 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   945 MBytes   793 Mbits/sec                sender
[  4]   0.00-10.00  sec   939 MBytes   787 Mbits/sec                receiver

-------------------------------------------------------------------------

iperf3.exe -c 192.168.1.1 -R
Connecting to host 192.168.1.1
Reverse mode, remote host 192.168.1.1 is sending
[  4] local 127.0.0.1  connected to 192.168.1.1
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec   318 KBytes  2.59 Mbits/sec
[  4]   1.01-2.01   sec   276 KBytes  2.26 Mbits/sec
[  4]   2.01-3.00   sec   276 KBytes  2.27 Mbits/sec
[  4]   3.00-4.01   sec   301 KBytes  2.44 Mbits/sec
[  4]   4.01-5.01   sec   242 KBytes  1.98 Mbits/sec
[  4]   5.01-6.01   sec   304 KBytes  2.49 Mbits/sec
[  4]   6.01-7.01   sec   272 KBytes  2.23 Mbits/sec
[  4]   7.01-8.01   sec   272 KBytes  2.23 Mbits/sec
[  4]   8.01-9.00   sec   272 KBytes  2.23 Mbits/sec
[  4]   9.00-10.01  sec   311 KBytes  2.53 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.01  sec  2.88 MBytes  2.42 Mbits/sec  995             sender
[  4]   0.00-10.01  sec  2.78 MBytes  2.33 Mbits/sec                  receiver

OpenWrt Ipv6-S2022.02.01  Sirpdboy 5.10.94

ChenLing93, 2022-02-12:
这个问题貌似我也遇到有

den5is5, 2022-02-12:
I downloaded a bunch of builds and this problem is everywhere.

den5is5, 2022-02-12:
installed 5.4.162 OpenWRT Ipv6-S2021.12.09 by Sirpdboy the same. Clean firmware is fine. one of the programs cuts the speed

ChenLing93, 2022-02-13:
这个问题是在哪里  可以告诉一下么

den5is5, 2022-02-13:
no one wants to solve the problem

den5is5, 2022-02-13:
I wrote to them in the chat, they didn't answer!

den5is5, 2022-02-15:
was able to raise the speed of 300 Mbits/sec

jason0316, 2022-02-15:
> was able to raise the speed of 300 Mbits/sec

how do u solve the poor speed?

den5is5, 2022-02-15:
SQM QoS,  Basic Settings set the speed of your card

ChenLing93, 2022-02-16:
问题找到了就好  我学习一下

den5is5, 2022-02-17:
> Как только проблема будет найдена, я изучу ее

it helped me. I hope it will help you too

den5is5, 2022-02-17:
> 问题找到了就好 我学习一下

it helped me. I hope it will help you too

closed 2025-05-14 by sirpdboy