sirpdboy/openwrt #340: BCM5719网卡能识别但不加载

shenghuang147, 2022-05-25:
# BCM5719网卡没有被加载

从 `3.1` 版本 全盘重刷到 `5.15` 版本

网卡 `BCM5719` 使用 `lspci` 能找到网卡 但是 `ifconfig` 中没有 `BCM5719` 如下图：

[image]

[image]

另外我发现在刷完固件第一次启动会出现 `can't open /etc/board.json no such file` 字样 如下图:

[image]

经过查询这好像与网络的初始化有关，不知道与我的问题有没有关系；

niceran, 2022-05-29:
我测试5.26版是可以的正常使用的，检查一下你的卡在其他系统能不能工作

shenghuang147, 2022-05-29:
> 

经过查询，可能是缺少驱动 `kmod-gt3`

现在在学习准备用官方源码自己编译一个

用别人编译好是限制太多了

shenghuang147, 2022-06-02:
@sirpdboy 感谢；

我提出的问题我已经测试完了

经过测试 `6.1` 版本 `BCM5719` 正常识别

closed 2022-06-02 by shenghuang147