sirpdboy/openwrt #349: iKuai虚拟机运行openwrt旁路由，设置旁路由网关的树莓派无法外网访问NAT，求解？

Leoin1, 2022-06-09:
固件版本【OpenWrt Ipv6-Plus-S20220601 by Sirpdboy】
宿主机：iKuai虚拟机运行此固件
issues：iKuai主路由为10.0.0.1，openwrt为10.0.0.223，树莓派设置网关为10.0.0.223时，正常上网，但是无法远程访问，在iKuai做的穿透。openwrt作为旁路由和iKuai互指。

树莓派4B的网络配置如下：
Change Mode    : [STATIC]                                                           
│                        Copy           : Copy current address to "Static"                                   
│                        Static IP      : [10.0.0.19]                                                        
│                        Static Mask    : [255.255.255.0]                                                    
│                        Static Gateway : [10.0.0.223]                                                       
│                        Static DNS     : [10.0.0.223 10.0.0.19 10.0.0.1 10.0.0.222]

xxx.com:1234 访问，iKuai映射到内网10.0.0.19:80，外网wan:1234 无法访问。

Leoin1, 2022-06-12:
遇到这个问题的同仁，可以看下这个issuses
https://github.com/Dreamacro/clash/issues/432

Leoin1, 2022-06-12:
遇到这个问题的同仁，可以看下这个issuses
https://github.com/Dreamacro/clash/issues/432
都是开启clash的

sirpdboy, 2022-06-13:
访问 是你爱快设置的事。。和OP无关

457992195, 2022-10-28:
@Leoin1 我也是这个问题，找到一个解决办法。
在爱快流控分流-端口分流中，设置一个分流方式为下一跳网关，线路选择op的ip，协议任意，原地址设置需要使用代理的ip，其他不用填。这样需要代理的地址既走了代理，端口映射也不需要在op中设置啥了

457992195, 2022-10-28:
> 

之前使用OpenClash的Redir-Host模式，只需要在主路由端口映射就可以，然后群辉、unraid或者服务器网关和dns填op的ip，在外网下直接使用域名+端口就可以访问的。后来更新OpenClash后，就不行了。

Leoin1, 2022-11-16:
>  @Leoin1 我也是这个问题，找到一个解决办法。
> 
> 在爱快流控分流-端口分流中，设置一个分流方式为下一跳网关，线路选择op的ip，协议任意，原地址设置需要使用代理的ip，其他不用填。这样需要代理的地址既走了代理，端口映射也不需要在op中设置啥了
> 
> 
> 
> 
> 
> 

试试在防火墙里加自定义设置转发 
我是这样解决的

```

iptables -t nat -I PREROUTING -p tcp -m multiport --sports 80 8080 -j RETURN

```

是你要转发的端口

Leoin1, 2023-02-22:
> > 
> 
> 之前使用OpenClash的Redir-Host模式，只需要在主路由端口映射就可以，然后群辉、unraid或者服务器网关和dns填op的ip，在外网下直接使用域名+端口就可以访问的。后来更新OpenClash后，就不行了。

`iptables -t nat -I PREROUTING -p tcp -m multiport --sports 80 8080 -j RETURN
#这里的端口一行最大只能填写15个
#下一跳地址以及ikuai设置我这边不生效
`

Leoin1, 2023-02-22:
> 访问 是你爱快设置的事。。和OP无关

刚看到大大回复，感谢  回去研究下ikuai设置

susuq233, 2025-01-12:
> > @Leoin1 我也是这个问题，找到一个解决办法。
> > 在爱快流控分流-端口分流中，设置一个分流方式为下一跳网关，线路选择op的ip，协议任意，原地址设置需要使用代理的ip，其他不用填。这样需要代理的地址既走了代理，端口映射也不需要在op中设置啥了
> 
> 试试在防火墙里加自定义设置转发 我是这样解决的
> 
> ```
> 
> iptables -t nat -I PREROUTING -p tcp -m multiport --sports 80 8080 -j RETURN
> ```
> 
> 是你要转发的端口

爱你，（づ￣3￣）づ╭❤～，卡了2天了，用这条命令解决了，感谢，呜呜呜！！！

closed 2025-05-14 by sirpdboy