sirpdboy/openwrt #359: 新版本建议

griphin, 2022-06-23:
感谢Sirpdboy大神的辛苦付出，让现在的固件版本越来越完善了。

如果后续版本，能做如下调整，那就更加完美了。

1. 精简版添加时下非常流行的istore插件，这就是openwrt内部的APPStore，插件安装卸载更加方便了。
2. 翻墙插件太多了，passwall, passwall2, shadowsocks plus, helloworld, bypass这些插件按理只需要运行一个，而且也看不出功能上的差别，感觉没有必要都给编译进去
3. Samba4文件共享，我觉得应该放在“存储”的菜单下，放到“管控”的菜单下，显得让人摸不着头脑
4. Docker最好能单独成为一个菜单项，这个应用还是很频繁的，每次在服务菜单下，总要花好几秒才能找到
5. Zerotier， NPS, FRP，放到VPN菜单下，感觉更加符合逻辑。
6. 固件要是支持在线更新就好了。

再次感谢Sirpdboy的付出，期待固件日臻完美。

shu0802, 2022-06-26:
> 感谢Sirpdboy大神的辛苦付出，让现在的固件版本越来越完善了。
>
> 如果后续版本，能做如下调整，那就更加完美了。
>
> 1. 精简版添加时下非常流行的istore插件，这就是openwrt内部的APPStore，插件安装卸载更加方便了。
> 2. 翻墙插件太多了，passwall, passwall2, shadowsocks plus, helloworld, [...]

只要互不干扰那就没有关系。

griphin, 2022-06-27:
> 第2点很自私

什么自私？

zhd-love, 2022-06-28:
建议将“向导”放入系统文件夹中。

sirpdboy, 2022-06-29:
插件之间冲突所以将有些插件放到其他位置，不信的你试试别人 的大全版，网络共享，DDNS，网络共享等 点击后会导致菜单 隐藏等 BUG。目前我这固件没有这个BUG了。至于什么 istore这个没开源，不会集成。samba4网络共享和存储中的 网络挂载 冲突 所以放管控中。其它的菜单 要改源码，分组暂时就目前这样处理。

ogr1432, 2022-07-08:
第二点不能苟同，一般整合到op系统内的梯子固件相对稳定，集成依赖，编译者考虑到位，用户按需选择，几个组件也不会增加很多固件体积或者磁盘空间，不启用更是对于资源占用影响可忽略不计；反而如果不集成靠用户自己后期安装很可能由于缺少依赖等技术问题导致失败或者使用不稳定因素，单纯以为功能重复就建议发布者去掉，既不利于固件覆盖更多用户，也不便不同需求的用户使用，说自私可能稍微有点过，但确实欠考虑。

griphin, 2022-07-08:
> 第二点不能苟同，一般整合到op系统内的梯子固件相对稳定，集成依赖，编译者考虑到位，用户按需选择，几个组件也不会增加很多固件体积或者磁盘空间，不启用更是对于资源占用影响可忽略不计；反而如果不集成靠用户自己后期安装很可能由于缺少依赖等技术问题导致失败或者使用不稳定因素，单纯以为功能重复就建议发布者去掉，既不利于固件覆盖更多用户，也不便不同需求的用户使用，说自私可能稍微有点过，但确实欠考虑。
==================================================================================
不扣帽子，毕竟文革结束都46年了。 咱就就事论事，梯子的功能如果除了翻墙，还有其他特殊用途，又或者，某些特殊的服务器，需要特定的插件，譬如有的服务器必须要用Bypass，而有的需要用到Passwall等等，那确实需要保留各种插件，我承认我眼界小了，让你们贻笑大方；但是如果Passwall, Shadowsocks, bypass, helloworld都可以，而且也没有什么区别，保留4个的意义在那里？更何况，很多人用的是刷Openwrt的硬路由，诸如新3，总共就32M空间，怎么可能给你整合那么多插件，这群用户不也用得挺好么。

closed 2022-07-08 by griphin