sirpdboy/openwrt #364: 关于I225网卡问题

winner890302, 2022-06-30:
6.15固件运行3天， 网口从2.5g掉到100M，  附上日志，  麻烦大神看下， 这是225网卡的毛病还是我家网线问题。。。。。。

Mon Jun 27 10:18:28 2022 kern.info kernel: [251177.020369] igc 0000:01:00.0 eth0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
Mon Jun 27 10:18:28 2022 kern.info kernel: [251177.028744] br-lan: port 1(eth0) entered blocking state
Mon Jun 27 10:18:28 2022 kern.info kernel: [251177.034062] br-lan: port 1(eth0) entered forwarding state
Mon Jun 27 10:18:28 2022 daemon.notice netifd: Network device 'eth0' link is up
Mon Jun 27 10:18:28 2022 daemon.notice netifd: bridge 'br-lan' link is up
Mon Jun 27 10:18:28 2022 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Jun 27 10:24:25 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 2 (fd 58) from 192.168.5.153 port 50992 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:24:25 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 57) from 192.168.5.153 port 50991 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:24:47 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 2 (fd 56) from 192.168.5.153 port 50993 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:24:47 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.5.153 port 50988 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:24:47 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 52) from 192.168.5.153 port 50989 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:24:47 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 53) from 192.168.5.153 port 50990 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:37 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 54) from 192.168.5.153 port 51728 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:59 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 56) from 192.168.5.153 port 51729 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:59 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 55) from 192.168.5.153 port 51727 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:59 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 2 (fd 52) from 192.168.5.153 port 51678 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:59 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.5.153 port 51676 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:26:59 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 53) from 192.168.5.153 port 51677 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:44:10 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 10:44:10 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 10:55:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.5.153 port 53040 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:55:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 54) from 192.168.5.153 port 53201 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:55:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 53) from 192.168.5.153 port 53041 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:56:19 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 56) from 192.168.5.153 port 53203 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:56:19 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 55) from 192.168.5.153 port 53202 is idle for more than 60 seconds - closing it.
Mon Jun 27 10:56:19 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 52) from 192.168.5.153 port 53042 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:26:20 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 56) from 192.168.5.153 port 53760 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 53) from 192.168.5.153 port 53711 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 2 (fd 54) from 192.168.5.153 port 53759 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 55) from 192.168.5.153 port 53761 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 52) from 192.168.5.153 port 53710 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:39:57 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 51) from 192.168.5.153 port 53709 is idle for more than 60 seconds - closing it.
Mon Jun 27 11:58:23 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 11:58:23 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 12:00:00 2022 cron.err crond[12547]: USER root pid 9747 cmd sync && echo 3 > /proc/sys/vm/drop_caches
Mon Jun 27 12:00:00 2022 kern.info kernel: [257269.641237] ash (9747): drop_caches: 3
Mon Jun 27 12:30:01 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 13:00:01 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 13:10:21 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 13:40:21 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 14:10:21 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 14:20:49 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 14:47:15 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 14:47:15 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 15:14:42 2022 user.info mwan3track[5122]: Check (ping) failed for target "119.29.29.29" on interface wan (pppoe-wan)
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.5.153 port 56222 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 4 (fd 55) from 192.168.5.153 port 56289 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 2 (fd 56) from 192.168.5.153 port 56288 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 54) from 192.168.5.153 port 56287 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 3 (fd 52) from 192.168.5.153 port 56221 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 daemon.info netdata[6111]: POLLFD: LISTENER: client slot 5 (fd 53) from 192.168.5.153 port 56220 is idle for more than 60 seconds - closing it.
Mon Jun 27 15:17:45 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 15:47:46 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 16:00:00 2022 cron.err crond[12547]: USER root pid 26893 cmd sync && echo 3 > /proc/sys/vm/drop_caches
Mon Jun 27 16:00:01 2022 kern.info kernel: [271669.471095] ash (26893): drop_caches: 3
Mon Jun 27 16:30:01 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 16:54:05 2022 daemon.notice ttyd[8378]: HTTP / - 192.168.5.153
Mon Jun 27 16:54:05 2022 daemon.notice ttyd[8378]: HTTP /token - 192.168.5.153
Mon Jun 27 16:54:05 2022 daemon.notice ttyd[8378]: WS   /ws - 192.168.5.153, clients: 1
Mon Jun 27 16:54:05 2022 daemon.notice ttyd[8378]: started process, pid: 9910
Mon Jun 27 16:54:12 2022 daemon.notice ttyd[8378]: WS closed from 192.168.5.153, clients: 0
Mon Jun 27 16:54:12 2022 daemon.notice ttyd[8378]: killing process 9910 with signal: 1 (SIGHUP)
Mon Jun 27 16:54:12 2022 daemon.notice ttyd[8378]: process killed with signal 1 (SIGHUP), pid: 9910
[... cut ...]
Mon Jun 27 21:37:56 2022 kern.info kernel: [291944.436438] br-lan: port 1(eth0) entered disabled state
Mon Jun 27 21:37:56 2022 daemon.notice netifd: Network device 'eth0' link is down
Mon Jun 27 21:37:56 2022 authpriv.notice beardropper[4773]: loadState() loaded 0 entries
Mon Jun 27 21:37:57 2022 daemon.notice netifd: bridge 'br-lan' link is down
Mon Jun 27 21:37:57 2022 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Jun 27 21:38:02 2022 kern.info kernel: [291951.143528] igc 0000:01:00.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX

winner890302, 2022-06-30:
这是内核日志

[251177.020369] igc 0000:01:00.0 eth0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
[251177.028744] br-lan: port 1(eth0) entered blocking state
[251177.034062] br-lan: port 1(eth0) entered forwarding state
[257269.641237] ash (9747): drop_caches: 3
[271669.471095] ash (26893): drop_caches: 3
[286068.356650] ash (12008): drop_caches: 3
[291944.431096] igc 0000:01:00.0 eth0: NIC Link is Down
[291944.436438] br-lan: port 1(eth0) entered disabled state
[291951.143528] igc 0000:01:00.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX

winner890302, 2022-07-01:
> i225跟你一个德行，用爱快做主路由几小时都掉成100m，用OP做主路由能坚持久点，但是经常断流。

哎。。 。

junke, 2022-08-01:
看来是硬件的通病了

closed 2025-05-14 by sirpdboy