sirpdboy/openwrt #393: upnp插件bug

GFDCGFC, 2022-08-20:
[image]
要这样才能修复

sirpdboy, 2022-09-07:
UPNP的这个问题是普遍现象？

GFDCGFC, 2022-09-07:
是普遍现象，要按图片的操作才能正常
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年9月7日 10:31 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [sirpdboy/openwrt] upnp插件bug (Issue #393) |

UPNP的这个问题是普遍现象？

sirpdboy, 2022-09-07:
这内容不太好修改，问题是你这修复的什么问题？ 现在用测试是没问题的哦。

GFDCGFC, 2022-09-07:
修改UPNP插件配置文件
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年9月7日 13:38 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [sirpdboy/openwrt] upnp插件bug (Issue #393) |

这内容不太好修改，问题是你这修复的什么问题？ 现在用测试是没问题的哦。

GFDCGFC, 2022-09-07:
不修改配置文件UPNP会开启失效
|
|
|

|
|

|
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2022年9月7日 13:38 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [sirpdboy/openwrt] upnp插件bug (Issue #393) |

这内容不太好修改，问题是你这修复的什么问题？ 现在用测试是没问题的哦。

sirpdboy, 2022-10-13:
10.13号发了 升级upnp，加入外网接口和内网地址二个参数，并且自动设置。   你试试。

closed 2023-03-03 by sirpdboy