sirpdboy/openwrt #425: cups 管理员账号密码不对

chuan1127, 2022-11-16:
我使用的是11月1号的版本，但是cups密码不对，按照说明Usename：root
password: rootpassword (先设置好 root 的密码)
进不去

sirpdboy, 2022-11-17:
亲测试没问题的哦。  你驱动不对吧。

chuan1127, 2022-11-17:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    

    font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    管理员那里进不去. 提示密码不对. 还没有到添加打印机那一步. 附件为操作的视频！请大佬查看！但是你的邮箱有限制，不能收取超过15M的邮件,  能否提供一个别的邮箱，我将视频发送给你！
    
        
            
        
    
    
     
    
        
                    ***@***.***
        
    
 

  
    ---- 回复的原邮件 ----
  
  
    
      
        
         发件人 
        
        
            ***@***.***>
            
        
      
      
        
         发送日期 
        
        
        2022年11月17日 18:34
        
      
      
        
         收件人 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         抄送人 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         主题 
        
        
              Re: [sirpdboy/openwrt] cups 管理员账号密码不对 (Issue #425)
        
      
    
  
  

亲测试没问题的哦。  你驱动不对吧。

closed 2022-11-25 by sirpdboy