sirpdboy/openwrt #430: 重新安装unblockneteasemusic失败

feilongproject, 2022-11-21:
想着想要重置unblockneteasemusic设置，就想着卸载重新安装，结果在安装的时候报错，请问怎么解决？（或者有没有适合的ipk？）
```
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libstdcpp6 and libstdcpp6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libatomic1 and libatomic1) providing same name marked HOLD or PREFER. Using latest.
Package luci-app-unblockneteasemusic (3.1-2) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-conntrack-netlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.79-1-53ea5a0694b1d3faa99c939dd703b18d) for kmod-ipt-ipset
```

sirpdboy, 2022-11-25:
重新刷机 或者 重置一下系统。 是卸载 里面的内核 重新下载就行，不是卸载 这个插件。

closed 2022-11-25 by sirpdboy

feilongproject, 2022-11-25:
> 重新刷机 或者 重置一下系统。 是卸载 里面的内核 重新下载就行，不是卸载 这个插件。

只能重新刷机 或者 重置一下系统么？数据要怎么备份