sirpdboy/openwrt #479: 建议修复一下roofts版在lxc容器下出现dnsmasq无法工作问题

mitsuxia, 2023-05-20:
建议修复一下roofts版在lxc容器下出现dnsmasq启动不了问题，openwrt查看后台服务发现53端口dnsmasq无法工作，重启服务也无济于事

Asurin219, 2023-10-14:
我也被这个问题困扰了很久，不过我看了下大神的帖子，注释掉/etc/init.d/dnsmasq中的以下字段可以解决这个问题：
```
# procd_add_jail dnsmasq ubus log
# procd_add_jail_mount $CONFIGFILE $DHCPBOGUSHOSTNAMEFILE $DHCPSCRIPT $DHCPSCRIPT_DEPENDS
# procd_add_jail_mount $EXTRA_MOUNT $RFC6761FILE $TRUSTANCHORSFILE
# procd_add_jail_mount $dnsmasqconffile $dnsmasqconfdir $resolvdir $user_dhcpscript
# procd_add_jail_mount /etc/passwd /etc/group /etc/TZ /etc/hosts /etc/ethers
# procd_add_jail_mount_rw /var/run/dnsmasq/ $leasefile
# case "$logfacility" in */*)
#       [ ! -e "$logfacility" ] && touch "$logfacility"
#       procd_add_jail_mount_rw "$logfacility"
# esac
# [ -e "$hostsfile" ] && procd_add_jail_mount $hostsfile
```

参考：[https://www.right.com.cn/forum/thread-8276823-1-1.html](https://www.right.com.cn/forum/thread-8276823-1-1.html)

clion007, 2024-05-24:
修改系统文件感觉始终不是好的方式，可能将相关的配置文件放到它默认的/tmp/dnsmasq.d路径下是现在的唯一合适的选择

closed 2025-05-14 by sirpdboy