sirpdboy/openwrt #491: 当开启nwan3多播后 负载均衡不生效 且一直保存中

vistal8, 2023-10-23:
路径 /cgi-bin/luci/admin/network/mwan/policy 设置后一直是保存中。J1900直装不正常 PVE不正常 但2022年版本可以 只是负载有些高   J4205 直装和PVE都不正常   换过浏览器 一样，其他都可以保存，但是这个保存不会生效 重启无效，pass不可以用，即使可以保存 但是TCP一直是未运行  
[image]
麻烦 大神修复一下  非常感谢

vistal8, 2023-10-24:
设置指定IP走指定多播口 也不生效  已经尝试过2023年的所有版本 均有此类问题

vistal8, 2023-10-24:
Active ipv4 user rules:
    0     0 - JD  all  --  *      *       0.0.0.0/0            192.168.2.0/24       
    0     0 - JD2  all  --  *      *       0.0.0.0/0            192.168.2.0/24       
    0     0 - jd3  all  --  *      *       0.0.0.0/0            192.168.2.0/24       
    0     0 - PC  all  --  *      *       0.0.0.0/0            192.168.2.0/24       
  862 61084 - balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            
    0     0 S https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 443

iovejieba, 2024-02-19:
> 设置指定IP走指定多播口 也不生效 已经尝试过2023年的所有版本 均有此类问题

去找OpenWrt Ipv6-Plus-R2022.11.01     mwan3正常  均衡正常 指定出口正常   实测已主力路由 一年