sirpdboy/openwrt #530: 插件能稳定点吗，配置好的内容一更新都没了

Dominic-010, 2025-02-16:
希望能稳定下插件内容，去除哪些插件可以有个提示

avijynisc2, 2025-02-25:
最搞笑的是   固件在线更新各种报错，我用DEEPSEEK查看居然他的是代码写错了。牛啊

Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_system/ota_flashing'.
Error while parsing template '/usr/lib/lua/luci/view/admin_system/ota_flashing.htm':
Syntax error in /usr/lib/lua/luci/view/admin_system/ota_flashing.htm:13: ')' expected near '='

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^   
这是报错原文。。。

原始代码片段
Copy<%

	local sys  = require "luci.sys"
	local fs = require "nixio.fs"
	if  fs.access('/var/run/ota/MD5_File') then
	     ip = sys.exec("head -2 "/var/run/ota/MD5_File" | awk -F "=" '{print $2}'")
	end

-%>
主要问题
在 Lua 中，字符串需要使用相同类型的引号来包围。您在 sys.exec 函数中使用了双引号来包围命令，而命令内部又使用双引号，导致语法冲突。您可以使用单引号来包围文件路径，或者将整个命令用单引号包围来避免这种情况。

修改后的代码
以下是修正后的代码：

Copy<%

	local sys  = require "luci.sys"
	local fs = require "nixio.fs"
	if fs.access('/var/run/ota/MD5_File') then
	     ip = sys.exec("head -2 /var/run/ota/MD5_File | awk -F '=' '{print $2}'")
	end

-%>

sirpdboy, 2025-02-26:
这个代码是准备扩展提示IP用，后来没用了。更新的时候 有一个版本 结果上传错了文件 。 这好笑吗？  每一个次写新功能都是经过无数的测试，一个符号不对都无法正常使用。这做为我们这种业余爱好者来说， 是要投入非常巨大的时间 和工作量。 希望大家能支持与理解 。

closed 2025-04-06 by sirpdboy