sirpdboy/openwrt #548: 后台管理界面ui 错误，以及最新版更新后 卡在开机界面

ByteSecurity, 2025-03-11:
自从 [Vip-Super-2503062120-6.6.75](https://github.com/sirpdboy/openwrt/releases/tag/Vip-Super-2503062120-6.6.75) 版本开始，后台就一直是这样报错红色提示， 并且所有文字都是英文，伴随功能加载不出来

`TypeError
_(...).format is not a function
`
[image]

尝试ota更新到最新， 但是卡在开机界面 ，更新失败。

辛苦了~

sirpdboy, 2025-03-12:
清除缓存和历史记录，不要保留配置升级。

ByteSecurity, 2025-03-13:
[image]

清除缓存没用，依然存在问题；

sirpdboy, 2025-03-13:
才新出的版本了，针对这个问题做了修正。你再试试。

ByteSecurity, 2025-03-30:
UI问题 依然存在， 每次清除缓存记录第一次进没问题，  同一个按钮多点几次 又有了；  

[image]

ByteSecurity, 2025-03-30:
不敢OTA 了， 各种安装的软件(python,npm)各种脚本 ota下 全没了。。

closed 2025-08-19 by sirpdboy