sirpdboy/openwrt #566: 主题页面切换的时候突然报错，

taotieren, 2025-04-12:
```bash

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/kucat/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: attempt to index local 'config' (a boolean value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: in main chunk

In error(), file [C]
  called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `            die(ex);`
  Near here --------^

```

sirpdboy, 2025-04-14:
是什么机器，什么版本。请明一下你的具体情况 。报错过程 。能重现的才是问题。

taotieren, 2025-04-14:
PLUS 里面的最新版本 x86_64 PVE 环境。在显示里面几个功能切换之间突然出现。zerotier-one 的显示还有问题，这个还没修复吗？参考之前 issue 里面修改配置文件能修复。

taotieren, 2025-04-22:
补充
切换到实时连接 流量 窗口又复现了 PLUS 最新版
```
http://openwrt.lan/cgi-bin/luci/admin/status/realtime/bandwidth

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/kucat/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: attempt to index local 'config' (a boolean value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: in main chunk

In error(), file [C]
  called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^

```

taotieren, 2025-04-22:
补充
在连接监控页面看连接数也会崩溃
```bash
http://openwrt.lan/cgi-bin/luci/admin/status/realtime/connections

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/kucat/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: attempt to index local 'config' (a boolean value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/kucat/header...."]:43: in main chunk

In error(), file [C]
  called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^

```

loso3000, 2025-07-20:
> 测试 是没有安装进阶设置的时候有这个可能，新版本已经做了这个处理，不再有这个问题了。

closed 2025-07-24 by sirpdboy